Octanis3/Octanis3-Electronics #2: Assemble & Test v1 PCB

traffael, 2017-03-05:


traffael, 2017-07-05:
Done testing:
- general microcontroller functionality/flashing/blink/...
- SPI communication between uC and RFID chip
- RFID communication (UID readout from card) with read range up to 5 cm with smartcard
- on/off IR LED and photodetector

TO DO:
- determine good intensity of IR LED and practical readout voltages of the phototransistor
- test read range with real bird ring & adjust matching network if needed
- Power supply (DC-DC converter from battery voltage to 3.3V)

In general, v1 seems to be usable for testing in the real context!

deragent, 2017-11-03:
So after some more extended testing things do not look so good anymore:
- Light Barrier was changed to use a  TSOP38238 for the receive side. This works very stable now in our Lab tests!
- The RFID communication was implemented and worked in general, but the system is very unstable and after some time (minutes) leads to system resets. Often after these resets the RFID chip stops working altogether. It often starts working again after re-soldering it or just heating it. We do not really know the source of these problems.
- The read range using bird tags is also very disappointing when it works, being on the order of 2-3 cm within the axis of the antenna, and basically 0 cm when not in the axis of the antenna.
- The DC/DC circuit is not working correctly at the moment. Given that the design was previously used in other projects this is most likely due to assembly problems (soldering etc).

The current strategy is to make some prototypes usable just having the light-barrier and storing the passages locally.
We also started working on a v2 prototype using 125kHz RFID instead of 13.65MHz.

traffael, 2019-04-06:
done

closed 2019-04-06 by traffael