Octanis3/Octanis3-Electronics #5: Assembly of second light barrier

traffael, 2017-03-05:
(could use the same v1 PCB and only mound phototransistor & LED and connect it to the fully mounted reader PCB)

deragent, 2017-03-05:
I would actually propose the second light barrier directly on the same PCB but on the other side! 
So the same position as the other light barrier but just on the lower side of the PCB. In order to not get interference between the two light barriers we can also switch the diode and the transistor on one side.

Given that this version is a prototype, I think it would be a good idea to put this directly in the design. We can still add a connector and some 0-Ohm resistors to disable this second light barrier and use an external one.
But I think to detect the passing and direction of a bird the proposed solution with two barriers just next to each other should work.

traffael, 2017-03-06:
have added a second phototransistor on the backside (it is presumably enough to have just one light source). the thickness of the PCB is probably not enough though to detect a difference in time when one or the other barrier was interrupted.

closed 2019-04-06 by traffael